kriswiner/ESP32 #1: images

kriswiner, 2016-12-14:
[image]

kriswiner, 2016-12-16:
[image]

kriswiner, 2016-12-16:
[image]

muktillc, 2017-10-27:
Hi Kris,

   one quick question regarding the SPI. I am trying to get the SPI work and my sensor is responding but for some reason the MISO pin (pin #19) is not accepting any input. I tested the output of the sensor and it gives the expected output for the WHO_AM_I register but as soon as I connect the SDO of the sensor to MISO, the output goes down to zero and I don't see anything oscilloscope. Wondering if you have seen anything like this.

Could not comment on your code so msging you here.
Thanks,
Bhaven.

kriswiner, 2017-10-27:
No idea, I haven't tried to use SPI on the ESP32. This sounds like
something for the ESP forum maybe?

Which dev board are you using?

In any case, the SPI pins for VSPI are MOSI 23, MISO 19, CLK 18, and 5 for
nCS althout you could probably use any pin for chip select.GPIO5 is usually
pulled up though.

Not sure why this is not working for you.