kriswiner/ESP32 #3: Dev Board Question

mjs513, 2017-02-17:
Hi Kris.

First you did it again.  Great board design.  Right now I have been experimenting with the ESP32 thing from Sparkfun and seem to be having i2c problems reading from a lsm9ds0 from adafruit.  I have not tested other sensors but was wondering if this dev board would have the same issues.  I have tested the sensor on a teensy 3.5 so I know it does work.

THanks
Mike

kriswiner, 2017-02-17:
I haven't had any issues with I2C, I have tried MPU9250, BME280, VEML6040, BNO055, etc. Lots of different ones and haven't had any trouble since the Arduino API was fixed about two months ago.

mjs513, 2017-02-17:
Hi Kris

Thanks for the info.  Think I am going to be ordering your version today.

Mike

closed 2017-02-25 by mjs513

kriswiner, 2017-02-25:
Hi Mike,

I missed this til just now. The ESP32 is still pretty immature even though they claim the wifi sector is nearly the same as the ESP8266. I think it is still going to be a while until we can use the ESP32 with the same degree of confidence and success we now enjoy with the ESP8266.

If the sketch is well commented I will post it on my ESP32 repository.

mjs513, 2017-02-25:
Having major problems getting it to work consistently.  The code works sometimes. Like you said still immature.  I also having problems with my 8266 code now. Not sure what's going on.  Was working now its not.  I for your board yeeaterday and just posted it as an issue. Cnat get anything to load on it.