kriswiner/ESP32 #4: Can not upload sketches

mjs513, 2017-02-25:
Hi Kris

Just soldered up your board and tried to upload a test sketch.  I keep receiving the following error messages:
A fatal error occurred: Failed to connect to ESP32: Invalid head of packet ('V')
A fatal error occurred: Failed to connect to ESP32: Invalid head of packet ('V')

I selected the ESP32 Dev Board from the Arduino IDE.  Is this the correct board.  I keep trying to upload but can not?

mjs513, 2017-02-25:
Ok.  Figured it out by reading the dragonfly getting started page.  You have to hold down the boot button and then press and release the reset button and then it will load the sketch.  Then press the reset button again to get into run mode. 

Just to let you know it connects in a fraction of a second and runs like a charm unlike the ESP32 Thing from Sparkfun which I could not get to connect for the life of me and communicate correctly. Now the sketch I have working with the associated processing code works like a charm.  

I will put it together for you and maybe you can give it a try with your sample 9150_ms56937 code.  I don't have that config boarded anywhere so I can not do it for you.  

Mike

kriswiner, 2017-02-25:
Yes, unlike the Dragonfly, the ESP32 always has to be put into boot mode to flash a new program, and then reset in order to run it on my board. Other boards have an auto reset so the last step isn't necessary.

Glad you got it to work.

Kris

closed 2017-02-25 by kriswiner

mjs513, 2017-02-25:
Thanks Kris.  PS I just ordered your MPU9250_MS5637 (mini).  Once I get it I will use your code and merge with my UDP code and post it for your.  I have it working with my FreeIMU code so I don't think it will be a problem with yours :) Thanks for all your work.  

Mike

kriswiner, 2017-02-25:
I look forward to it! Thanks.