kriswiner/ESP32 #5: WiFi Example:

mjs513, 2017-03-04:
Hi Kris

I got your board today and hooked it up.  Tested with my modified code and it worked fine.  

I am attaching two zip files.  One with a modified MPU9250 sketch and a second with the associated Processing sketch. You will need the Arduino ESP32 library.  I did modify boards.txt file to include your board.  Basically its the ESP32 Dev Board with DFU substituted for DIO.   Let me know how it works for you.  
[FreeIMU_cube_UDP_ES32.zip](https://github.com/kriswiner/ESP32/files/818327/FreeIMU_cube_UDP_ES32.zip)
[MPU9250_MS5637_AHRS.zip](https://github.com/kriswiner/ESP32/files/818328/MPU9250_MS5637_AHRS.zip)

Here is what I added:

```
esp32t.name=ESP32 Dev Module Onehorse

esp32t.upload.tool=esptool
esp32t.upload.maximum_size=1044464
esp32t.upload.maximum_data_size=294912
esp32t.upload.wait_for_upload_port=true

esp32t.serial.disableDTR=true
esp32t.serial.disableRTS=true

esp32t.build.mcu=esp32
esp32t.build.core=esp32
esp32t.build.variant=esp32
esp32t.build.board=ESP32_DEV_Onehorse

esp32t.build.f_cpu=240000000L
esp32t.build.flash_mode=dout
esp32t.build.flash_size=4MB

esp32t.menu.FlashFreq.80=80MHz
esp32t.menu.FlashFreq.80.build.flash_freq=80m
esp32t.menu.FlashFreq.40=40MHz
esp32t.menu.FlashFreq.40.build.flash_freq=40m

esp32t.menu.UploadSpeed.921600=921600
esp32t.menu.UploadSpeed.921600.upload.speed=921600
esp32t.menu.UploadSpeed.115200=115200
esp32t.menu.UploadSpeed.115200.upload.speed=115200
esp32t.menu.UploadSpeed.256000.windows=256000
esp32t.menu.UploadSpeed.256000.upload.speed=256000
esp32t.menu.UploadSpeed.230400.windows.upload.speed=256000
esp32t.menu.UploadSpeed.230400=230400
esp32t.menu.UploadSpeed.230400.upload.speed=230400
esp32t.menu.UploadSpeed.460800.linux=460800
esp32t.menu.UploadSpeed.460800.macosx=460800
esp32t.menu.UploadSpeed.460800.upload.speed=460800
esp32t.menu.UploadSpeed.512000.windows=512000
esp32t.menu.UploadSpeed.512000.upload.speed=512000

esp32t.menu.DebugLevel.none=None
esp32t.menu.DebugLevel.none.build.code_debug=0
esp32t.menu.DebugLevel.error=Error
esp32t.menu.DebugLevel.error.build.code_debug=1
esp32t.menu.DebugLevel.warn=Warn
esp32t.menu.DebugLevel.warn.build.code_debug=2
esp32t.menu.DebugLevel.info=Info
esp32t.menu.DebugLevel.info.build.code_debug=3
esp32t.menu.DebugLevel.debug=Debug
esp32t.menu.DebugLevel.debug.build.code_debug=4
esp32t.menu.DebugLevel.verbose=Verbose
esp32t.menu.DebugLevel.verbose.build.code_debug=5

```

mjs513, 2017-03-04:
I do have a question on your example.  It looks like you calculate the madgwick algorithm every ten time and then output every so often based a set interval.  Why do both?

mjs513, 2017-03-04:
Aquick question. Can I use a 3.3v lipo attached directly to vbat and gneiss? Also what's the Max size on the lipo I can use?

kriswiner, 2017-03-04:
No max size, but V < 5.5 V. VBAT is the same as battery Anode. But there is a JST port on top left for a LiPo battery which is usually more convenient.

kriswiner, 2017-03-04:
The madgwick and mahony algorithms are iterative. they asymptote to a stable orientation estimtion so you get best results by iterating multiple times on every new data sample. How many times is up to you (or the speed of your processor).

mjs513, 2017-03-04:
I guess you are talking about the + pin and the unlabeled pin? Otherwise I think I am mission.g something.

mjs513, 2017-03-04:
Thanks for explaining your iterative approach. What I found was you really see that effect on the algorithm startup and when you stop moving the sensor.

mjs513, 2017-03-04:
Have fun with wifi piece. Let me know if you need more comments.

kriswiner, 2017-03-04:
Yes, the small two-pin port is for a JST connector, + is the LiPo anode.

mjs513, 2017-03-04:
Got it thanks.

closed 2017-04-26 by mjs513