kriswiner/ESP32 #9: failed to load bootloader header!  error with new esp32 lib/ide 1.8.2

mjs513, 2017-04-28:
Getting error message using ide 1.8.2 and new version of esp32 library. Something about magic number.  I reported under issue 334, https://github.com/espressif/arduino-esp32/issues/334

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:2036
load:0x40078000,len:9988
load:0x40080000,len:252
entry 0x40080034
E (30) esp_image: image at 0x1000 has invalid magic byte
E (30) boot: failed to load bootloader header!
user code done
```

mjs513, 2017-04-28:
Going to go ahead and close this since you are on the other thread.

mjs513, 2017-05-06:
Hi Kris,

Reopened since the new ESP32 core seems to be conflicting with the board.  Its opened as Issue [#342](https://github.com/espressif/arduino-esp32/issues/342) on the ESP32 GitHub site.  Not sure if you are seeing this same problem.

Mike

mjs513, 2017-12-19:
Hi Kris,

Think this is now fixed, at least with the latest changes on GitHub.  But I am running into another problem, not sure if it is associated with the bootloader.  I press the boot button and press and then release the reset button.  Everything seems to upload no errors.  But when I press and release the rst button again to put in run mode the sketch doesn't seem to run.  Tested it on my sketch, then the blink sketch with ledpin on 5 and a couple of others

Any idea?

kriswiner, 2017-12-19:
Yes, the latest core fixes whatever problems we were having flashing new
sketches.

I haven't run into this problem before. Are you sure the device is out of
boot mode? Check the voltage on SDA (GPIO0), it should be 3V3 for normal
running. I am assuming you checked the obvious like using enough power,
turning on and off, etc.

Presumably it used to work; now all of a sudden it doesn't? Did you choose
4 MByte flash, DOUT for flashing?

mjs513, 2017-12-19:
> Did you choose 4 MByte flash, DOUT for flashing?
Thanks for reminding me about this.  They added your board to dropdown list of boards in boards managed.  So I went back in and checked to see the setup for the board.  4MB was correct, but it was set at DIO.  I changed it DOUT and it reset into run mode and the blink sketch started working.  Will give the other sketches a try to make sure they work and will let you know.  I will post an issue on the esp32 GitHub so they get it fixed once I check it all out.

Thanks

mjs513, 2017-12-20:
Hi Kris,

I am beginning to think that the issue is with I2C with the ESP32 Core.  I extracted and ran the following code on a T3.5 and it worked fine.  It identified the WhoIAm correctly.  Same code on the ESP32 board failed WhoIAm test.  I did a little digging and it is almost reminiscent of [Issue #90](https://github.com/espressif/arduino-esp32/issues/90) that you opened up on the Arduino-ESP32 core library.  There was another issue with I2C but it was fixed - https://github.com/espressif/arduino-esp32/issues/834

```
#include "Wire.h"   

#define MPU9250_ADDRESS 0x68  // Device address when ADO = 0
#define WHO_AM_I_MPU9250 0x75 // Should return 0x71

void setup() 
{
  Serial.begin(115200);
  delay(4000);

  Wire.begin(400000); //(SDA, SCL) (21,22) are default on ESP32, 400 kHz I2C clock
  delay(1000);

  // Set up the interrupt pin, its set as active high, push-pull
  pinMode(5, OUTPUT);
  digitalWrite(5, LOW);

  I2Cscan();// look for I2C devices on the bus
  delay(1000);

  // Read the WHO_AM_I register, this is a good test of communication
  Serial.println("MPU9250 9-axis motion sensor...");
  uint8_t c = readByte(MPU9250_ADDRESS, WHO_AM_I_MPU9250);  // Read WHO_AM_I register for MPU-9250
  Serial.print("MPU9250 "); Serial.print("I AM "); Serial.print(c, HEX); Serial.print(" I should be "); Serial.println(0x71, HEX);
  
}

void loop() {
  // put your main code here, to run repeatedly:

}

// simple function to scan for I2C devices on the bus
void I2Cscan() 
{
    // scan for i2c devices
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++ ) 
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();

    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");

      nDevices++;
    }
    else if (error==4) 
    {
      Serial.print("Unknown error at address 0x");
      if (address<16) 
        Serial.print("0");
      Serial.println(address,HEX);
    }    
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");
}

  uint8_t readByte(uint8_t address, uint8_t subAddress)
{
  uint8_t data = 0;                        // `data` will store the register data   
  Wire.beginTransmission(address);         // Initialize the Tx buffer
  Wire.write(subAddress);                  // Put slave register address in Tx buffer
  Wire.endTransmission(false);             // Send the Tx buffer, but send a restart to keep connection alive
  Wire.requestFrom(address, 1);  // Read two bytes from slave register address on MPU9250 
  data = Wire.read();                      // Fill Rx buffer with result
  return data;                             // Return data read from slave register
}

```

kriswiner, 2017-12-20:
> was fixed - espressif/arduino-esp32#834

Don't you have to specify which pins you want to use for I2C with the ESP32?

mjs513, 2017-12-20:
I did that when I ran it on the ESP32.  The code I posted was the T3.5 version.  Sorry, should have been clearer.

kriswiner, 2017-12-20:
So you are saying you don't get the correct result on the ESP32? I haven't
tried for a while but last time I did it worked fine...

mjs513, 2017-12-20:
Yep.  Just changed the board to the esp32 dev board so I could change the core debug level and here is the error for reading the whoiam register:

> MPU9250 9-axis motion sensor...
> [W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 68
> [W][esp32-hal-i2c.c:334] i2cRead(): Ack Error! Addr: 68
> MPU9250 I AM FF I should be 71

kriswiner, 2017-12-20:
I'll try it later today...

mjs513, 2017-12-20:
I am using a git pull from a few days ago but didn't see anything in the new commits that would affect I2C.

kriswiner, 2017-12-20:
I am always reluctant to use the latest core just for this reason... Mine
is a pull from two months ago or so..

mjs513, 2017-12-20:
Just as a FYI I just tested the BNO055 and that appears to work as well as the BME280.  As for the pull - was looking for the link to latest release but for reason couldn't find it so I used the latest core pull.

kriswiner, 2017-12-20:
How could the BNO055 work but the MPU9250 not? This doesn't make any kind
of sense...

kriswiner, 2017-12-20:
Sure you have pullups on the MPU9250 board?

mjs513, 2017-12-21:
For your board no pullups (thought there were onboard 4.7k's), on the DIYMALL 9250 board there are 4.7k pullups on the sda/scl lines already.  Both are behaving exactly the same.

kriswiner, 2017-12-21:
Yes, you are right, no sensor board pullups required. Maybe the DIYMall
board is junk...

mjs513, 2017-12-21:
Tried both boards with a T3.5 and they worked - along with a couple different versions of a 9250 library.  That's one of the first thing I tried.  Going to add 4.7k's to your board and see what happens I have the mini version and it looks like I am suppose to solder the back side to activate the 4.7k's?

kriswiner, 2017-12-21:
Shouldn't need them for my Mini MPU9250 board, but if you want them then it
is the top dolder jumper that need to be soldered. You do have solder the
back if you want to supply 3V3 and GND from the side rather than top PTHs.

Did you verify that on the DIYMall MPU9250 board SDA and SCL show 3V3?

I just don't understand how the BNBO055 will ACK on a general address call
but the MPU9250 won't. Doesn't make sense...

mjs513, 2017-12-21:
You may want to check this issue out: https://github.com/espressif/arduino-esp32/issues/834

Maybe makes more sense to you.

kriswiner, 2017-12-21:
> You may want to check this issue out: espressif/arduino-esp32#834
>
> Maybe makes more sense to you.

wtf!

I thought I2C was "fixed" on the ESP32 but this looks like a can of rotten
worms. I guess this validates my decision to wait a few months before even
trying to use the ESP32 for anything serious.

I do know that I had to use the clock stretching on the ESP8285 for reading
the CAM M8Q via I2C. But other than that, I haven't encountered this wierd
problem (or at least for a long time) with the espressif wire API

mjs513, 2017-12-21:
Yeah, I know I saw you issue #90 for the ESP32.  It looks like they changed the wire API about a month or so ago to try and resolve more issues, but looks like broke it at least for the 9250 or 6050 (will have to test it later).  Think I read in the thread that they are saying that those boards are special cases :( - yeah I know.  Think I should go ahead and generate an issue for them :)

Have to dig up my old version of the core - think I have it on a hard drive somewhere.

kriswiner, 2017-12-21:
If the MPU9250 and MPU6050 are Wire.h special cases then the ESP32 system
layer is in worse shape than I thought...

I fear the ESP32 won't be ready for prime time for a while yet...

mjs513, 2017-12-21:
Ok.  Dug out my old copy from around Feb of this year works fine for the 9250.

kriswiner, 2017-12-21:
If you are going to post an issue, maybe you could suggest they go back to
this previous version and un-fix (fubar) the Wire.h library.

mjs513, 2017-12-21:
Planning on it.  Was just waiting to here your opinion. :)

mjs513, 2017-12-21:
Ok here is the issue I posted, if you want to add for it: https://github.com/espressif/arduino-esp32/issues/937

kriswiner, 2017-12-21:
Well done!

Nothing for me to add here.

It will be a good test of Espressif's commitment to the ESP32 Arduino IDE
since with this kind of buggy API the ESP32 is basically useless.

mjs513, 2017-12-21:
Hope me-no-dev works this.  He seems to have a good handle on what's going on.

kriswiner, 2017-12-21:
Yes, he fixed Wire.h last time. Too bad they chose to fix it again!

mjs513, 2017-12-22:
Think I isolated to when the problem started - it was when they incorporated the commit for long I2C reads.  I posted what I found so maybe that is a starting point for them.

mjs513, 2017-12-25:
See issue 90 for problem and how to fix it.  Have to read 2 bytes.  Was your original issue back in Dec 2016.

Merry christmas.

kriswiner, 2017-12-25:
I thought they fixed this...

mjs513, 2017-12-25:
Kris.  From what I can see it was fixed until they added support for long i2c addressimg. On a hunch  I went ahead and tried the two byte read in issue 90 and it worked. There long addressing broke it.  Now it's up to them to fix it I guess. Just wanted to let you know.

kriswiner, 2017-12-25:
Thanks Mike. I forgot about this issue and it was fun to read what was
going on a year ago. Sad that they are still having trouble with the
basics. As far as I know BLE doesn't work well yet either. I guess creating
a functioning Arduino core for the ESP32 is harder than people expected,
but I haven't been using the ESP32 since this initial bit of work as I have
been waiting for a bit of maturity, but apparently it is still not ready
for prime time. I hope they get there soon though, since the concept is a
good one.

Anyway, I hope you told espressif too, since I would expect me-no-dev could
fix it again somehow...

mjs513, 2017-12-25:
Yes I did in the same issue that I opened up.  I referenced issue 90 (even gave them the link) and mentioned that me-no-dev fixed it.  They will get around to it eventually.  Think they are working on a new i2c/wire wrapper - me-no-dev is involved with that one.

Can't wait for the ble support for the Arduino library.  Like you said still immature in a lot of respects.  Can't blame them - there's a lot there.