kriswiner/ESP32 #12: Esp32 compiling guide

carbonadam, 2018-01-13:
I can not seem to get these sketches to compile.  I am using arduino IDE 1.8.5 and a doit esp32.  i can read i2c devices on the i2c pins fine but I get the following when trying compile the BNO055_MS5637_ESP32 Basic Example Code.  

C:\Users\woodtiger\Documents\Arduino\libraries\ESP32\Bosch\BNO055_MS5637\BNO055_MS5637.ino: In function 'void loop()':

BNO055_MS5637:557: error: 'MadgwickQuaternionUpdate' was not declared in this scope

   MadgwickQuaternionUpdate(ax, -ay, -az, gx*PI/180.0f, -gy*PI/180.0f, -gz*PI/180.0f,  -my,  -mx, -mz);

                                                                                                     ^

C:\Users\woodtiger\Documents\Arduino\libraries\ESP32\Bosch\BNO055_MS5637\BNO055_MS5637.ino: In function 'void magCalBNO055(float*)':

BNO055_MS5637:933: error: narrowing conversion of '32768' from 'int' to 'int16_t {aka short int}' inside { } [-Wnarrowing]

   int16_t mag_max[3] = {0x8000, 0x8000, 0x8000}, mag_min[3] = {0x7FFF, 0x7FFF, 0x7FFF};

                                               ^

BNO055_MS5637:933: error: narrowing conversion of '32768' from 'int' to 'int16_t {aka short int}' inside { } [-Wnarrowing]

BNO055_MS5637:933: error: narrowing conversion of '32768' from 'int' to 'int16_t {aka short int}' inside { } [-Wnarrowing]

C:\Users\woodtiger\Documents\Arduino\libraries\ESP32\Bosch\BNO055_MS5637\BNO055_MS5637.ino: In function 'uint8_t readByte(uint8_t, uint8_t)':

C:\Users\woodtiger\Documents\Arduino\libraries\ESP32\Bosch\BNO055_MS5637\BNO055_MS5637.ino:1142:29: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:

  Wire.requestFrom(address, 1);            // Read one byte from slave register address 

                             ^

In file included from C:\Users\woodtiger\Documents\Arduino\libraries\ESP32\Bosch\BNO055_MS5637\BNO055_MS5637.ino:37:0:

C:\Users\woodtiger\Documents\Arduino\hardware\espressif\esp32\libraries\Wire\src/Wire.h:65:13: note: candidate 1: uint8_t TwoWire::requestFrom(int, int)

     uint8_t requestFrom(int, int);

             ^

C:\Users\woodtiger\Documents\Arduino\hardware\espressif\esp32\libraries\Wire\src/Wire.h:63:13: note: candidate 2: uint8_t TwoWire::requestFrom(uint8_t, uint8_t)

     uint8_t requestFrom(uint8_t, uint8_t);

             ^

exit status 1
'MadgwickQuaternionUpdate' was not declared in this scope

kriswiner, 2018-01-13:
Yeah, you need to add the extra filequaternionfilters.ino as an additional
file in the tabs, or just comment out MagwickQuaternion in the main sketch.

carbonadam, 2018-01-14:
Fantastic thank you

closed 2018-01-14 by carbonadam