kriswiner/ESP32 #13: Yaw resets to previous value

carbonadam, 2018-01-15:
In your example Im having some odd values with the software Yaw,  if it reads 90 and I move it to 180 it will count back down to 90 again..So no matter where I rotate the sensor it resets to what ever value it starts with
The hardware Yaw seems to keep its values however.

kriswiner, 2018-01-15:
What are you using for a breakout board, what are your sensors?

carbonadam, 2018-01-15:
shamefully one from china,  BNO055 SW Revision ID: 3.11
[image]

kriswiner, 2018-01-15:
> [image:
> gy-bno055-9dof-9-axis-bno055-absolute-orientation-imu-ahrs-breakout-sensor-accelerometer-gyroscope-triaxial-geomagnetic]

The probelm is in my code, for some reason, the magnetometer data are not
updated properly unless using the AMG mode, I think, where just the sensor
data are returned. When the fusion engine is used these data seems to be
messed up, don;t know why. Just use the hardware solution, you don't need
the software solution unless you are doing the fusion on the host for some
reason. No reason to use both.

kriswiner, 2018-01-15:
BTW, that is one ugly board design!

carbonadam, 2018-01-16:
ok now that makes sense:)  I had thought for some reason the hardware values were not fusion values.  and yeah the board is ugly and Im not even sure what that other part is with EAUC.  Thanks a lot for your help hopefully I have a nice robot video soon and can share it with you:)))

kriswiner, 2018-01-16:
The big black part is a crystal for the emedded Cortex M0 MCU. Optional and
not really useful IMHO.