kriswiner/ESP32 #14: esp32 mpu9250

carbonadam, 2018-03-16:
Any chance for a version without the sensor hub? eek:)

kriswiner, 2018-03-16:
A version of what?

carbonadam, 2018-03-16:
A sketch for the mpu9250 for the esp32 using the Madgwick and/or Mahony filters

kriswiner, 2018-03-16:
https://github.com/kriswiner/ESP8285/tree/master/MPU9250

close enough

carbonadam, 2018-03-16:
I managed to get this one of yours to work 
https://github.com/kriswiner/MPU9250/blob/master/MPU9250BasicAHRS.ino

Though its not written specifically for the esp32.  I wonder how this compares to the sensor fusion hub output?

kriswiner, 2018-03-16:
4 degree heading accuracy rather than 2 degree with the EM7180.

carbonadam, 2018-03-16:
Outstanding:)  Any increases in accuracy  if I was to use the floating point available on the esp32?  Preferring this for the moment over the bno55.

kriswiner, 2018-03-16:
Accuracy will depend more on how well you calibrate the sensor, but if you
can run the fusin algorith at >> 1 kHz this can only help.

The BNO055 is not a good solution.

carbonadam, 2018-03-16:
Right Calibration.  Any tips to calibration using the code version I used?

kriswiner, 2018-03-16:
Use the code version I linked to.

carbonadam, 2018-03-16:
Hmm, compiling that seems to give me a bunch of problems with the librarys..any hints?

kriswiner, 2018-03-16:
Put the quaternion filter file in the main sketch tab?

carbonadam, 2018-03-16:
more todo with the ESP8266WebServer.h: and queue.h..file not found..even though I copied in those libs

kriswiner, 2018-03-16:
Commen out the offending lines then.

carbonadam, 2018-03-17:
got as far as commenting out the offending temperature and pressure reading..it compiles but no sensor readings updating
ax = 0.00 ay = 0.00 az = 0.00 mg
gx = 0.00 gy = 0.00 gz = 0.00 deg/s
mx = 0 my = 0 mz = 0 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
Gyro temperature is 28.7 degrees C
Digital temperature value = 0.00 C
Digital temperature value = 32.00 F
Digital pressure value = 0.00 mbar
Altitude = 145366.45 feet
Yaw, Pitch, Roll: 13.80, 0.00, 0.00
Grav_x, Grav_y, Grav_z: 0.00, 0.00, 1000.00 mg
Lin_ax, Lin_ay, Lin_az: 0.00, 0.00, -1000.00 mg
rate = 158308.53 Hz
ax = 0.00 ay = 0.00 az = 0.00 mg
gx = 0.00 gy = 0.00 gz = 0.00 deg/s
mx = 0 my = 0 mz = 0 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
Gyro temperature is 28.7 degrees C
Digital temperature value = 0.00 C
Digital temperature value = 32.00 F
Digital pressure value = 0.00 mbar
Altitude = 145366.45 feet
Yaw, Pitch, Roll: 13.80, 0.00, 0.00
Grav_x, Grav_y, Grav_z: 0.00, 0.00, 1000.00 mg
Lin_ax, Lin_ay, Lin_az: 0.00, 0.00, -1000.00 mg
rate = 158305.33 Hz

kriswiner, 2018-03-17:
Use the calibration functions then in the first sketch that you got to work.