kriswiner/ESP32 #15: general esp32 / mpu question

razor950, 2018-03-21:
Hello

I have an esp32 from HiLetgo. I also have an MPU 9255. I want to know if the following log shows a working 9255. I mostly left it still on the table.

https://nopaste.xyz/?35c2cef5d570a4f6#rl1+EIpTHcsxTk90dUk3Zn9v7aqtTQD36r6SvTjkHDM=

kriswiner, 2018-03-21:
> 6SvTjkHDM=

looks OK, but did you move the sensor alla round during the mag
calibration? Should do.

razor950, 2018-03-21:
yes, I usually move it, I did not do it that time. 

I need to poke your head then, I've wanted to test this code and so I got the esp32 to spit out the quaternion values like he did. https://github.com/virtuellerealitaet/WearableIMU/

I then use his unreal project to get the rotating cube to rotate but it seems to randomly spew data massively when I don't move the device and if I move it it doesn't really register due to the amount of data being sent / or spammed in this case.

kriswiner, 2018-03-21:
Not a sensor problem...

razor950, 2018-03-21:
Scanning...
I2C device found at address 0x68  !
done

MPU9250 9-axis motion sensor...
MPU9250 I AM FF I should be 73
Could not connect to MPU9250: 0xFF

I am getting this now and I have tried a few different cables :(

kriswiner, 2018-03-21:
Are you using 0x68 as the MPU9250 address?

I don't quite understand since the last time everything was working. What
did you change?

razor950, 2018-04-02:
hey, sorry

I got it all working, my issue with the code I was having trouble with was that the delay of 10 was wrong when i2c clock speed is 400 kHz.