kriswiner/ESP32 #16: Problem gyro bias

exzibith, 2018-03-27:
hi kris,

on my last question on your thread issue https://github.com/kriswiner/MPU9250/issues/241
i used sketch MPU9250_MS5637_AHRS_t3.ino,

now i tried to used your sketch MPU9250_MS5637_AHRS.ino, link as below :
https://github.com/kriswiner/ESP32/tree/master/MPU9250_MS5637

my question:
are my sensor had broken? because i tried with ESP32 sketch i got something wrong with my gyro bias (dps) (i put sensor on flat surface and steady position), i tried calibration about 3 times, with same problem. can you help me or explain what should i do? 
are the gyro bias can affected the real time gyro data?

**_Test 1_**
Scanning...
I2C device found at address 0x0C  !
I2C device found at address 0x68  !
done

MPU9250 9-axis motion sensor...
MPU9250 I AM 73 I should be 73
MPU9250 is online...
x-axis self test: acceleration trim within : -0.2% of factory value
y-axis self test: acceleration trim within : 0.2% of factory value
z-axis self test: acceleration trim within : 0.9% of factory value
x-axis self test: gyration trim within : -0.4% of factory value
y-axis self test: gyration trim within : -0.3% of factory value
z-axis self test: gyration trim within : 0.3% of factory value
 Calibrate gyro and accel
accel biases (mg)
30.09
31.01
-18.98
gyro biases (dps)
**3.03**
0.00
-0.98
MPU9250 initialized for active data mode....
AK8963 I AM 48 I should be 48
AK8963 initialized for active data mode....
Mag Calibration: Wave device in a figure eight until done!
Mag Calibration done!
AK8963 mag biases (mG)
219.92
1031.42
-736.31
AK8963 mag scale (mG)
0.60
1.56
1.46
X-Axis sensitivity adjustment value 1.16
Y-Axis sensitivity adjustment value 1.17
Z-Axis sensitivity adjustment value 1.13

**_Test 2_**
Scanning...
I2C device found at address 0x0C  !
I2C device found at address 0x68  !
done

MPU9250 9-axis motion sensor...
MPU9250 I AM 73 I should be 73
MPU9250 is online...
x-axis self test: acceleration trim within : -0.3% of factory value
y-axis self test: acceleration trim within : 0.2% of factory value
z-axis self test: acceleration trim within : 0.8% of factory value
x-axis self test: gyration trim within : -0.5% of factory value
y-axis self test: gyration trim within : -0.2% of factory value
z-axis self test: gyration trim within : 0.2% of factory value
 Calibrate gyro and accel
accel biases (mg)
35.52
29.66
-13.92
gyro biases (dps)
**3.03**
0.10
-0.97
MPU9250 initialized for active data mode....
AK8963 I AM 48 I should be 48
AK8963 initialized for active data mode....
Mag Calibration: Wave device in a figure eight until done!
Mag Calibration done!
AK8963 mag biases (mG)
207.70
1012.09
-719.38
AK8963 mag scale (mG)
0.59
1.57
1.49
X-Axis sensitivity adjustment value 1.16
Y-Axis sensitivity adjustment value 1.17
Z-Axis sensitivity adjustment value 1.13

**_Test 3_**
Scanning...
I2C device found at address 0x0C  !
I2C device found at address 0x68  !
done

MPU9250 9-axis motion sensor...
MPU9250 I AM 73 I should be 73
MPU9250 is online...
x-axis self test: acceleration trim within : -0.3% of factory value
y-axis self test: acceleration trim within : 0.1% of factory value
z-axis self test: acceleration trim within : 0.7% of factory value
x-axis self test: gyration trim within : -0.5% of factory value
y-axis self test: gyration trim within : -0.2% of factory value
z-axis self test: gyration trim within : 0.3% of factory value
 Calibrate gyro and accel
accel biases (mg)
27.77
32.84
-17.46
gyro biases (dps)
**3.06**
0.06
-0.96
MPU9250 initialized for active data mode....
AK8963 I AM 48 I should be 48
AK8963 initialized for active data mode....
Mag Calibration: Wave device in a figure eight until done!
Mag Calibration done!
AK8963 mag biases (mG)
237.37
998.03
-722.77
AK8963 mag scale (mG)
0.60
1.33
1.70
X-Axis sensitivity adjustment value 1.16
Y-Axis sensitivity adjustment value 1.17
Z-Axis sensitivity adjustment value 1.13

kriswiner, 2018-03-27:
> on my last question on your thread issue kriswiner/MPU9250#241

What exactly is the problem?

After applying the gyro biases, what is your gyro data read when flat and
motionless?

kriswiner, 2018-03-27:
> on my last question, you tell me any something wrong with my gyro data.
>
> Gyro at rest (no motion) should be much less that 1
>
> are my result calibration is right?
>
> ax = -0.43 ay = -0.98 az = 1001.04 mg
> gx = 0.12 gy = -0.06 gz = -0.08 deg/s
> mx = -26 my = 64 mz = 51 mG
> q0 = 0.98 qx = 0.00 qy = -0.00 qz = -0.19
> Gyro temperature is 29.2 degrees C
> Digital temperature value = 20.00 C
> Digital temperature value = 68.00 F
> Digital pressure value = 0.00 mbar
> Altitude = 145366.45 feet
> Yaw, Pitch, Roll: 351.38, -0.02, 0.17
> Grav_x, Grav_y, Grav_z: -2.94, -0.27, 1000.00 mg
> Lin_ax, Lin_ay, Lin_az: 2.52, -0.71, 1.04 mg
> sumCount = 1010
> sum = 0.56
> rate = 1803.96 Hz
> ax = -2.44 ay = -2.01 az = 1001.83 mg
> gx = 0.11 gy = 0.02 gz = 0.02 deg/s
> mx = -28 my = 69 mz = 71 mG
> q0 = 0.98 qx = 0.00 qy = -0.00 qz = -0.20
> [... 68 lines cut ...]
> Grav_x, Grav_y, Grav_z: -7.28, 4.00, 999.97 mg
> Lin_ax, Lin_ay, Lin_az: 8.14, -3.63, -0.82 mg
> sumCount = 1010
> sum = 0.56
> rate = 1805.72 Hz

gx = 0.12 gy = -0.06 gz = -0.08 deg/s

This looks fine, what's the problem?

Your mag needs calibration now.

exzibith, 2018-03-27:
ok kris, thanks for evaluation.

> Your mag needs calibration now.

there are my value data after calibration. 
> gx = 0.12 gy = -0.06 gz = -0.08 deg/s

should i do for calibration mag again?
======================================

then i have other question on motionless position, are result YAW is big value?
but if i rotate or move i get YAW normal value again like as pitch & roll value

from my sample data:
> Yaw, Pitch, Roll: 351.38, -0.02, 0.17

> Yaw, Pitch, Roll: 350.75, -0.04, 0.07

> Yaw, Pitch, Roll: 348.58, -0.03, 0.02