kriswiner/ESP32 #19: Problem with Roll,pith and yaw

FilipeRG1, 2019-01-22:
Hello Kris,
Please excuse me for my English.

I'm trying to use your MPU9250 reading code, but unfortunately the roll, pitch and yaw angles are being printed as "nan".
Quaternions also have the same problem.

The print
ax = -563.48 ay = -931.58 az = 616.94 mg
gx = 12.00 gy = -7.16 gz = 88.43 deg/s
mx = 2147483647 my = 2147483647 mz = 2147483647 mG
q0 = nan qx = nan qy = nan qz = nan
Gyro temperature is 17.7 degrees C
Yaw, Pitch, Roll: nan, nan, nan
Grav_x, Grav_y, Grav_z: nan, nan, nan mg
Lin_ax, Lin_ay, Lin_az: nan, nan, nan mg
sumCount = 0
sum = 0.00
rate = nan Hz

kriswiner, 2019-01-22:
Magnetometer not working:

mx = 2147483647 my = 2147483647 mz = 2147483647 mG

FilipeRG1, 2019-01-23:
Do you think the problem is in the module?

kriswiner, 2019-01-23:
Likely software/code you are using.

FilipeRG1, 2019-01-23:
I tried with another MPU and got the following data

ax = 891.72 ay = 231.99 az = -97.84 mg
gx = 3.04 gy = -4.48 gz = -68.95 deg/s
mx = -93 my = -23 mz = -50 mG
q0 = -0.23 qx = 0.28 qy = 0.82 qz = 0.44
Gyro temperature is 29.2 degrees C
Yaw, Pitch, Roll: 174.42, -38.96, 129.96
Grav_x, Grav_y, Grav_z: -596.00, -628.81, -499.38 mg
Lin_ax, Lin_ay, Lin_az: 1487.73, 860.80, 401.54 mg
sumCount = 630
sum = 0.56
rate = 1133.86 Hz

Do you think these data are right?
and thank you very much your code will be of great use.

kriswiner, 2019-01-23:
Looks better, but when the sensor is flat and motion less you should get 0,
0 1000 mg on the accel and 0, 0, 0 dps on the gyro.

Mag should read such that sqrt(mx**2 + my**2 + mz**2) = 500 mG