kriswiner/ESP32 #20: Is esp32 fast enough for ahrs using magwick or ekf

shomedas, 2019-04-04:
I tried the esp32_5637 code on Arduino Nano. I am unable to get rid of drift. Is esp32 fast enough for sensor fusion at high frequencies? I need to synchronise the angle measurement with a 30fps camera

kriswiner, 2019-04-04:
Should be. You can also try the various STM32L4 devices here
<https://www.tindie.com/stores/TleraCorp/> if you don't need wifi and/or
want ultra-low-power. Most important consideration after sensor suite is
proper calibration.

shomedas, 2019-04-04:
I have calibrated the magnetometer using ellipsoid method. For accelerometer I have removed offset using your code. For gyroscope I used your 5637 code. But the gyro error was in tens of millidps