kriswiner/ESP32 #23: ESP32 only  MPU9250 without MS5637

andrelmbraga, 2020-06-15:
Dear Kriswiner,

Do you have any example with ESP32 + MPU9250 without MS5637 ? 

My mission is do a angle level where I need to get ate least roll and pitch values, from one machine, save, and after check if the other machine has the same values. 

I was doing with mpu6050 however when the operator turn the sensor in YAW sense, the pitch and roll values change. That is a big problem because they have some difficult to put exactly in the same postion before calibrate the machine.

With you example and using MPU9250 I guess I'll solve this problem.

kriswiner, 2020-06-15:
I am afraid you are going to have to manage this on your own.

andrelmbraga, 2020-06-16:
Its done, I have used your MPU9250BasicAHRS. Just a few changes in order to work. 

The output:

`14:38:43.685 -> MPU9250 I AM 71 I should be 71
14:38:44.694 -> MPU9250 is online...
14:38:45.487 -> x-axis self test: acceleration trim within : 1.0% of factory value
14:38:45.487 -> y-axis self test: acceleration trim within : 0.3% of factory value
14:38:45.521 -> z-axis self test: acceleration trim within : 2.7% of factory value
14:38:45.521 -> x-axis self test: gyration trim within : 46.1% of factory value
14:38:45.521 -> y-axis self test: gyration trim within : 7.8% of factory value
14:38:45.521 -> z-axis self test: gyration trim within : 0.8% of factory value
14:38:47.360 -> MPU9250 initialized for active data mode....
14:38:47.360 -> AK8963 I AM 48 I should be 48
14:38:48.403 -> AK8963 initialized for active data mode....
14:38:48.403 -> X-Axis sensitivity adjustment value 1.18
14:38:48.403 -> Y-Axis sensitivity adjustment value 1.19
14:38:48.403 -> Z-Axis sensitivity adjustment value 1.14
14:38:49.414 -> ax = -66.96 ay = 1.77 az = 933.72 mg
14:38:49.414 -> gx = -0.07 gy = 0.02 gz = -0.08 deg/s
14:38:49.414 -> mx = -323 my = 134 mz = -658 mG
14:38:49.414 -> q0 = 0.06 qx = 0.81 qy = -0.39 qz = 0.43
14:38:49.414 -> Yaw, Pitch, Roll: -74.15, -48.31, -159.46
14:38:49.414 -> rate = 0.15 Hz
14:38:49.902 -> ax = -67.69 ay = 2.08 az = 934.20 mg
14:38:49.902 -> gx = -0.08 gy = 0.05 gz = 0.02 deg/s
14:38:49.902 -> mx = -324 my = 132 mz = -639 mG
14:38:49.902 -> q0 = 0.92 qx = 0.03 qy = -0.06 qz = 0.38
14:38:49.936 -> Yaw, Pitch, Roll: 30.57, -7.37, 1.03
14:38:49.936 -> rate = 1062.95 Hz
14:38:50.425 -> ax = -63.84 ay = 0.98 az = 935.91 mg
14:38:50.425 -> gx = -0.14 gy = -0.02 gz = -0.01 deg/s
14:38:50.425 -> mx = -323 my = 148 mz = -644 mG
14:38:50.425 -> q0 = 0.88 qx = 0.01 qy = 0.01 qz = 0.47
14:38:50.425 -> Yaw, Pitch, Roll: 42.84, 0.47, 2.09
14:38:50.425 -> rate = 1065.36 Hz
14:38:50.913 -> ax = -64.51 ay = -0.79 az = 935.12 mg
14:38:50.913 -> gx = -0.15 gy = 0.02 gz = 0.03 deg/s
14:38:50.948 -> mx = -335 my = 132 mz = -670 mG
14:38:50.948 -> q0 = 0.85 qx = -0.00 qy = 0.02 qz = 0.53
14:38:50.948 -> Yaw, Pitch, Roll: 49.87, 1.83, 1.06
14:38:50.948 -> rate = 1066.25 Hz
14:38:51.436 -> ax = -67.14 ay = 0.49 az = 933.53 mg
14:38:51.436 -> gx = -0.13 gy = 0.12 gz = -0.02 deg/s
14:38:51.436 -> mx = -330 my = 141 mz = -644 mG
14:38:51.436 -> q0 = 0.83 qx = -0.01 qy = 0.02 qz = 0.56
14:38:51.436 -> Yaw, Pitch, Roll: 54.69, 2.65, 0.58
14:38:51.436 -> rate = 1068.53 Hz
14:38:51.946 -> ax = -64.39 ay = -0.24 az = 939.09 mg
`

Now I have some doubts about the mag calibration, I was looking in this issue:
https://github.com/kriswiner/MPU9250/issues/50

However the MPU9250BaiscAHRS is really different when I compare with the example that you give to us:
https://github.com/kriswiner/MPU-9250/blob/master/MPU9250_MS5637_AHRS_t3.ino

For example I dont even had this function magcalMPU9250() in the example.  And I found this lines, that I'm not sure, but I think they are bias values, as we can find in the gyro.

`   readMagData(magCount);  // Read the x/y/z adc values
    getMres();
    magbias[0] = +470.;  // User environmental x-axis correction in milliGauss, should be automatically calculated
    magbias[1] = +120.;  // User environmental x-axis correction in milliGauss
    magbias[2] = +125.;  // User environmental x-axis correction in milliGauss`

Could you just give me some tips How can I ajust for my own sensor?

kriswiner, 2020-06-16:
> kriswiner/MPU9250#50 <https://github.com/kriswiner/MPU9250/issues/50>

readMagData(magCount); // Read the x/y/z adc values getMres(); magbias[0]
= +470.; // User environmental x-axis correction in milliGauss, should be
automatically calculated magbias[1] = +120.; // User environmental x-axis
correction in milliGauss magbias[2] = +125.; // User environmental x-axis
correction in milliGauss
This is from calibrating my particular sensor, you should delete this or
pace your own calibration values here.

There are lots of examples of calibration functions that you can use.

andrelmbraga, 2020-06-16:
I have read some information about this calibrations, I'm not really sure if I understood the method. For sure, I got that we need to move the sensor in an "eight" format when it is acquiring the data

andrelmbraga, 2020-06-16:
Thank you, that's what I undestood reading your code. Now I'm worried how can I get my magbias[].

Using this function: 

`void magcalMPU9250(float * dest1, float * dest2)
{
uint16_t ii = 0, sample_count = 0;
int32_t mag_bias[3] = {0, 0, 0}, mag_scale[3] = {0, 0, 0};
int16_t mag_max[3] = {0xF000, 0xF000, 0xF000}, mag_min[3] = {0x7FFF, 0x7FFF, 0x7FFF}, mag_temp[3] = {0, 0, 0};

Serial.println("Mag Calibration: Wave device in a figure eight until done!");
delay(4000);

sample_count = 128;
for(ii = 0; ii < sample_count; ii++) {
MPU9250readMagData(mag_temp); // Read the mag data
for (int jj = 0; jj < 3; jj++) {
if(mag_temp[jj] > mag_max[jj]) mag_max[jj] = mag_temp[jj];
if(mag_temp[jj] < mag_min[jj]) mag_min[jj] = mag_temp[jj];
}
delay(135); // at 8 Hz ODR, new mag data is available every 125 ms
}

// Get hard iron correction
mag_bias[0] = (mag_max[0] + mag_min[0])/2; // get average x mag bias in counts
mag_bias[1] = (mag_max[1] + mag_min[1])/2; // get average y mag bias in counts
mag_bias[2] = (mag_max[2] + mag_min[2])/2; // get average z mag bias in counts

dest1[0] = (float) mag_bias[0]_MPU9250mRes_MPU9250magCalibration[0]; // save mag biases in G for main program
dest1[1] = (float) mag_bias[1]_MPU9250mRes_MPU9250magCalibration[1];
dest1[2] = (float) mag_bias[2]_MPU9250mRes_MPU9250magCalibration[2];

// Get soft iron correction estimate
mag_scale[0] = (mag_max[0] - mag_min[0])/2; // get average x axis max chord length in counts
mag_scale[1] = (mag_max[1] - mag_min[1])/2; // get average y axis max chord length in counts
mag_scale[2] = (mag_max[2] - mag_min[2])/2; // get average z axis max chord length in counts

float avg_rad = mag_scale[0] + mag_scale[1] + mag_scale[2];
avg_rad /= 3.0;

dest2[0] = avg_rad/((float)mag_scale[0]);
dest2[1] = avg_rad/((float)mag_scale[1]);
dest2[2] = avg_rad/((float)mag_scale[2]);

Serial.println("Mag Calibration done!");
}`

Should I use dest1 or dest2? or none of them ? in orderm to replace your values.  I ll try to adapt this calibration fucntion in your example MPU9250BaiscAHRS.  Another point is, I didnt find any place to set my coordenate (location). Is that alright?

kriswiner, 2020-06-16:
Look at more recent sketches for how to use this calibration function. Your
location doesn't matter, but your magnetic declination does.