kriswiner/ESP32 #24: Is a WHO_AM_I response of "0x70" a sure sign of a MPU6500?

LeTo37, 2021-02-16:
Hi!

Thanks a lot for your code, I hope you've had many a free beer because of it!

I am trying to use the "MPU9250_MS5637_AHRS.ino" code (minus the MS5637) and can't seem to get any type of meaningful interaction with the AK8963. My WHO_AM_I response for the AK8963 is 0xFF. Granted, I am using a cheaper board, however it looks a lot like the real deal, the placements and layout look to be the same, the chip has MP92 written on it and PIN1 (RESV) is connected to VDDIO (PIN8). Having said that, my MPU9250 WHO_AM_I response is 0x70 which I think is an indicator that the chip is actually an MPU6500?

Is there potentially something else I could check to be sure? Or is the response I'm getting a sure fire sign?

kriswiner, 2021-02-16:
0x70 is rather odd, but it is complicated. Invensense made several
different variants of the MPU9250 including the MPU9255 with extra
pedometry embedded as well as some variants just for the mass production
Chinese market, so you could have one of these. These usually use the
MPU9250 plastic body but lack the mag, so could be an MPU6500  or something
else/worse; this is why you might not see the magnetometer. Since the
MPU9250 is out of production, I wouldn't be surprised if manus of these
cheap boards are using whatever inventory they can find and marketing them
as "genuine" MPU9250. But if you paid <$5 for your board, I would say you
are getting  precisely what you paid for....

LeTo37, 2021-02-16:
Hey, thanks for the quick response!
Ah okay, it seems I must have gotten one of the mp92 plastic bodies with who knows what inside. Such is the chance you take with those boards. Do you have any suggestions for cheap, 9DOF IMU boards? I believe you sell some hardware?

kriswiner, 2021-02-16:
0x70 is definitely the MPU6500 WH_AM_I, so might still be ok if you just
need a 6DoF.

I would recommend something like this
<https://www.tindie.com/products/onehorse/all-st-motion-sensor-breakout-board/>
for even better sensor quality than the MPU9250.

The ST sensor suite or MPU9250 with a motion co-processor might be a better
choice.. Unless you want to spend time developing your own sensor fusion,
using the EM7180 or MAX32660 co-processors saves a lot of time.